ljalves/hfeasy #1: HTTP server closes connection

futuretap, 2019-02-04:
I've installed the v3 firmware on my Wifi plug. For whatever reason I can open the /status and /config web pages only in Chrome. Neither Safari on Mac, nor Safari on iPhone nor Chrome on iPhone nor a generic HTTP client work. They all complain that the remote side has closed the connection.

However, setting the switch using /state?sw= works fine from all browsers. Any idea?

ljalves, 2019-02-13:
The webserver itself is builtin the SDK - not sure if I can do anything about that.
Does the module wifi config webpage have the same issue? (http://ip/)

futuretap, 2019-02-13:
No that works just fine. Only /config and /status are affected.

ljalves, 2019-02-13:
Confirmed... installed Safari for windows and it doesn't open those pages... weird!
I'll try to debug it.

Firefox and chrome in windows do work ok...

ljalves, 2019-02-14:
Fixed on commit ca4d996c4ae90511600d881e4ebae702406b24d5
(available on version 0v4)

closed 2019-02-14 by ljalves

futuretap, 2019-02-14:
Thank you, works great!