ljalves/hfeasy #3: MQTT publish

lordkolio, 2019-04-20:
Hello,

First thanks a lot for the good work with this switch! What is the command topic to turn on or off the switch with mqtt?

Thanks a lot.

ljalves, 2019-04-20:
Go to http://module_ip/config
There you can setup the MQTT server, topics and ON/OFF commands.

renzo56789, 2019-05-29:
Hello, i think it has my same problem...MQTT is connectd and tnks a lot man! that's crazy!
but i and maybe hime have doubts about configuration

MY REC:
Subscribe topic: hfF0FE6BD2EDBE
Publish topic: rec1
QOS: 0
ON value: 1
OFF value: 0

on home assistant:
switch:
- platform: mqtt
  name: "cucina"
  icon: mdi:lightbulb-on
  command_topic: "cmnd/rec1/POWER"
  state_topic: "stat/rec1/POWER"
  payload_on: "ON"
  payload_off: "OFF"
  retain: true

but it doesn't work

ljalves, 2019-05-30:
Tty changing the configs to:

REC:
Subscribe topic: cmnd/rec1/POWER
Publish topic: stat/rec1/POWER
QOS: 0
ON value: 1
OFF value: 0

home assistant:
platform: mqtt
name: "cucina"
icon: mdi:lightbulb-on
command_topic: "cmnd/rec1/POWER"
state_topic: "stat/rec1/POWER"
payload_on: "1"
payload_off: "0"
retain: true

renzo56789, 2019-05-31:
I solved it , i just have to follow http commands 

but if the lamp in ON and i restart HA it shows lamp OFF so i have to click ON to allineate status. It seems cause by retain set to 0 instead 1:
SWITCH STATE WEB PAGE
ret=0, sw='none' relay_state=1

page http://192.168.1.15/state

can i change that value ?

ljalves, 2019-06-01:
If you want to use the http response from the status page, you can look at the value returned by "relay_state": 0 = off, 1 = on

renzo56789, 2019-06-01:
what means this ret value on this page?
SWITCH STATE WEB PAGE
ret=0

lordkolio, 2019-06-12:
> what means this ret value on this page?
> SWITCH STATE WEB PAGE
> ret=0

Is it possible to show your config for http request? :)

ljalves, 2020-02-24:
> 
> 
> what means this ret value on this page?
> SWITCH STATE WEB PAGE
> ret=0

ret is just for debug, it just tells if the page was requested with the "?sw=.." parameter.
I will clean this page on the next release.

Closing.

closed 2020-02-24 by ljalves