ljalves/hfeasy #5: LED flickering

korgak, 2019-07-31:
The blue LED on my wifi smart plug keeps going on and off. Can we disable it? Because that is really annoying and unnecessary in a bedroom.

Malmbergs
Wi-fi smart plug
Model: SW6201

ljalves, 2019-08-04:
Whats the flicker frequency?
It should only flicker on and off when it is not connected to the server.
Once the connection is established, it should "calm down"...

korgak, 2019-08-04:
Thanks for the response. It is once each 2 or 2.5 seconds and continuously. I can switch the device nonetheless so it is still connected to the wifi.

ljalves, 2020-02-24:
It flickers when disconnected from the mqtt server (not wifi).
I will add some options to select the wifi LED behaviour.
Maybe these options for the LED:
- OFF
- MQTT status (as is right now).
- HTTP status (blink when being controlled by http requests).
- ?

Any other suggestions?

korgak, 2020-02-25:
O that would be such an improvement, I think your suggestions are just fine.
Thanks a lot.

ljalves, 2020-03-10:
Enhancement available on release 0v6 (available on gdrive).

The LED is configurable on the main config page.
Also added an option to locate the module - the LED will flicker in an unique pattern".

closed 2020-03-10 by ljalves