ljalves/hfeasy #6: JSON Arduino ESP8266

chris83-dev, 2019-10-27:
Bonjour et merci bien pour ce formidable projet,
je pilote les prises chacon avec un esp8266 via une esquisse de ce style grace à la librairie httpclient  : 

void chacon(String chacon_state, String chacon_ip)
  {
        WiFiClient client = server.available();
        HTTPClient http;

        String url = chacon_ip+"state?sw="+chacon_state;
        http.begin(url);
        //GET method
        int httpCode = http.GET();
        //String payload = http.getString();
        //Serial.println(url);
        //Serial.println(payload);
        http.end();
  }

je viens à vous pour savoir si vous savez comment
récupérer le statu de la prise avec un arduino et
si éventuellement vous pourriez intégrer le moyen
de récupérer le statu de la prise via une page web en JSON
merci bien, bonne soirée 
 Chris-M

ljalves, 2020-01-21:
English please.

ljalves, 2020-01-30:
Google translate tells me that you would like to receive the switch status in JSON format.
Let me see what I can do...

ljalves, 2020-02-24:
Webpage /state was changed to output json format.
Release 0v5

closed 2020-02-24 by ljalves