ljalves/hfeasy #7: Can't connect to mosquitto mqtt broker

msantagata, 2019-11-01:
Hi!
I have a RaspberryPi (OS: dietpi) with homebridge and mosquitto: both seem to work fine.
My problem is that I cannot get my Ankuoo rec modules to connect to the mqtt server...
Here are my configuration files:

MODULE
[image]

HOMEBRIDGE
[image]

mqtt settings seem to work - If I subscribe another mqtt client to those topics, I correcly see homebridge sending messages.
Anyway, in the status page of the ankuoo module I always get "MQTT server: Disconnected".
Any ideas on what I'm missing?

Thank you in advance, and congratulations for your excellent work on this firmware.

phrogg, 2020-01-21:
Same problem here.

ljalves, 2020-01-21:
Hi, It might be stupid but did you press the "Commit values" and then "Save to flash and reboot?"

msantagata, 2020-01-21:
Hi @ljalves , I promise I have :)
And the values do get saved, as I see them in the fields after reboot.
But in the status page I always get "MQTT server: Disconnected" even if I'm sure my mosquitto server is up and running, and the data I enter is correct.
Thanks!

phrogg, 2020-01-21:
For me the issue was that my password for mqtt was too long (64 Chars), now I have 25 chars and it works!

ljalves, 2020-01-21:
Interesting... never thought on that and now looking at the code I can see that I only have a max. of 30 chars for password... do you think I should make it bigger?

phrogg, 2020-01-21:
Hmh I think the length is okay, sure always is bigger better but 30 is good enough. On the other hand if it is possible to make it bigger than why not do it :) ?

polmdm, 2020-03-16:
Hi, I've flashed an Ankuoo NEO with PLUG UPGRADE. Then with MQTT, I've liked it to Home Assistant as a switch and works (I'm able to turn on and off). The issue is that when I press the switch, the light turn on but on HA the switch toggle off after few seconds. Then to turn it off I need to turn on again the switch and rapidly turn it off. It seems that the switch is unable to return the state to home assistant and for that reason it toggle off on Love UI.

My setup on the http://192.168.XX.YY/config_mqtt is:
Server IP: 192.168.XX.XX /
Server Port: 1883
usr: username
psw: ****
sub_topic: cmnd/abatjour/POWER
pub_topic: stat/abatjour/POWER
QOS: 0
ON value: 1
OFF value 0

The setup on Home Assistant configuration.yaml is:
switch:
  -platform: mqtt
    name: "light"
    command_topic: "cmnd/abatjour/POWER
    state_topic: "stat/abatjour/POWER
    payload_on: "1"
    payload_off: "0"
    retain: false

Actually I've made tons of attempts changing the value of qos, retain and tried adding other value to HA configuration like availability topic etc.

Someone can help me?

ljalves, 2020-03-16:
"Ankuoo NEO"? That plug is not on the "supported hardware" list... does it work?? (apart from the issue with the state).
What's the plug reference model number? SW5101?

About your issue, several things can be going wrong.
Check this:
1) Check the "timer" config page (http://ip/timer). Is there any OFF timer configured?
2) Check the "status" page  (http://ip/status):
   - Here you can also see the timer status (if it is waiting on something).
   - Then check the module uptime (something is causing a reset) and if it is check the reset reason.
   - Check the GPIO status "last changed" and see what caused the module to turn off.

Post here your results.

polmdm, 2020-03-17:
Hi, 
yes it worked, I've given it a try and it seems to work, the reference model should be hf-lpb100, it's the number on top left in web interface berfor flashing the software.

1. no there aren't any OFF timer configured
2. yes I used the status page to check if homeassistant is able to control the switch and it does:
     - both timers (on and off) are disabled
     - the uptime seems to work properly
     - the gpio status respect the reality (the light is on)

I'll attach a screenshot to be sure:
[image]

ljalves, 2020-03-17:
If this status page AFTER it turning OFF automatically?

If not, then you really need to check the status page AFTER the switch turns OFF (for no reason) to find out what caused it.

ljalves, 2020-03-17:
By the way, the reference model is printed in the sticker at the back (not asking for the wifi module model).
Should be something like: SW6201 or SW7201

ljalves, 2020-03-17:
By the way, this is more like the issues described on #9 
I'll close this one and re-open that issue - please reply here: https://github.com/ljalves/hfeasy/issues/9

closed 2020-03-17 by ljalves