ljalves/hfeasy #8: Support for Dimmer Wall Switch ?

amitn322, 2020-01-10:
Hi there,

Does this firmware support wall switch dimmer ? 

https://www.amazon.com/gp/product/B07C5YKRSG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I know that the model number is HF-LPB100 . I have access to the web UI . 

Thank you !

ljalves, 2020-01-21:
The current firmware doesn't support the dimmer switch.
Should be easy to add it but I don't own one to develop the code...

amitn322, 2020-01-21:
Any hints/pointers/instructions ? May be i could try to add that feature ? I have  some programming background.

ljalves, 2020-01-21:
Are you able to build the FW?
(Keil µVision with toolchain MDK-ARM)

I could provide you sample code to control that dimmer.

amitn322, 2020-01-21:
Yes I could build the FW I believe. 

Thank You

ljalves, 2020-01-21:
I will add basic support in the code, then you can compile, test and fix whatever needed.

amitn322, 2020-01-21:
Sounds good, thanks a bunch !

amitn322, 2020-01-29:
Hi there, 
were you able to add the support for dimmer ?

ljalves, 2020-01-29:
Yes... I've added it 8 days ago.
Check the commit history: https://github.com/ljalves/hfeasy/commits/master

Not sure if it is working as is (no way to test), but the basics are there.

ljalves, 2020-01-29:
BTW, if you aren't able to build the firmware maybe I can do it and then you do the testing.
The current code should make the dimmer switches (up/down) work.
If it does work, then all that is left is to add the mqtt control code.

amitn322, 2020-01-29:
Yeah Sorry, I saw it after I replied.. I was trying to download all the tools to build the firmware but if you already have the environment and can build it , that would be awesome. Also I am using hubitat so i am not sure mqtt is needed but other people might benefit.  Another question.. is there a way to backup existing firmware from the device ?

ljalves, 2020-01-29:
Flash this fw using the webpage:
https://drive.google.com/open?id=1aloEHXnd2Himrh6k5oSt7SN4Yn5hLhbn
And test if the buttons work (up/down).
If they do, then I'll finish the code to make it work with http requests and mqtt.
Let me know what are the results...

To revert back to original fw, flash this file:
https://drive.google.com/open?id=1VFUYzisX_9Dg3poMTeHph-QcMyM0k1bU

amitn322, 2020-01-29:
I did upload the firmware and the device rebooted ,but I still get the old interface and old firmware :(

ljalves, 2020-01-30:
That doesn't make sense.
Please remember that the main webpage is "hardcoded" (it will not change).
Can you try browsing to: [http://module_ip/status](http://module_ip/status) and report what you see?

amitn322, 2020-01-30:
This is what i see : 

`HF Easy v0.4 module status
System

Reset flags: 00000002, reason: softreset
Free memory: 25308 bytes
Uptime: 0 days 2h26m10s
GPIO status

Switch: High
Relay: Open(Off), last changed by: HTTP, publish: 0, action: 0
Timer status

Countdown (turn OFF): disabled
Countdown (turn ON): disabled
Connectivity
Hostname:
MQTT server: Disconnected`

amitn322, 2020-01-30:
Oh you know what, looks like it actually flashed :( my bad, the interface looked the same and i was expecting it to look a little different.. also my REC app is showing it offline now.

amitn322, 2020-01-30:
But I do not see any buttons like you mentioned :(

ljalves, 2020-01-30:
Yes, the main webpage is fixed, that's what makes these modules almost "unbrickable" (easy to flash back another fw and wifi settings are kept).

> But I do not see any buttons like you mentioned :(
I mean the physical buttons (in the wall).
Do they work?

amitn322, 2020-01-30:
Nope, they don't seem to work.. i can't even turn on/off the switch anymore..

ljalves, 2020-01-30:
Ok, bad news - try flashing the original FW and see if all works good again.
I will check the code and try to find bugs.

amitn322, 2020-01-30:
Does not seem to work with the original firmware that you provided either.

ljalves, 2020-01-30:
Hmm... that might explain why it didn't work...
Does the switch show in the App? If yes, there's a option to upgrade the FW in the app. Try that.

amitn322, 2020-01-30:
It shows offline in the app. I do have another switch as well, but i am not sure if there is a way to extract firmware from there.

ljalves, 2020-01-30:
Yes there is, but you have to open it up and connect the module serial port to a usb-serial adapter...
If you are able do that, can you post some pictures of the insides?
By the way, is the switch exactly as the one you posted on the first message?

amitn322, 2020-01-30:
Yes that is the exact switch, I don't think i would be able to extract the firmware with usb-serial adapter unfortunately :(

ljalves, 2020-01-30:
I find it weird that the app doesn't show the switch. Did you try a "factory reset"?
Also, are you able to open up the switch and take some photos of the board?

amitn322, 2020-01-30:
Okay, I will try to pull it out and take some pictures tomorrow. 

Thank You

ljalves, 2020-01-30:
Btw, dont do a factory reset - we might loose acess to webpage due to wifi getting unconfigured!
I have some ideas to make it work. Let me do some tests on my side first.
Meanwhile if you can get pictures of the insides it may help.

amitn322, 2020-01-30:
Actually I did do factory reset, but i am still able to access webpage over 10.x address..I will get pictures tomorrow.

ljalves, 2020-01-30:
1 question: did you try the physical buttons to turn on/off the lights after flashing the original firmware that I provided?

amitn322, 2020-01-30:
Yes I did try that and it does not work.

ljalves, 2020-01-30:
I'm uploaded another version. When you have the chance try flashing this one:
https://drive.google.com/open?id=19bs4oJEwXHVaf-VLe3xM--cLO4pOrPAY

Please test physical buttons and app.

amitn322, 2020-01-30:
I uploaded this version and now i don't have the "DEFAULT Open WIFI" .. The physical buttons or app does not work.

amitn322, 2020-01-30:
Nevermind, the default wifi came back after some time.. the physical buttons still don't work.. no LEDs blinking on the switch.

ljalves, 2020-01-30:
This means that they have 2 different dimmer modules...
I'll try to find out more info on that...
Meanwhile, whenever you can, disassemble your module and take some detailed board pictures - it should help.

ljalves, 2020-01-30:
Whats the name of the app you use with those dimmers?

amitn322, 2020-01-30:
REC Smart by Ankuoo on google play store.

amitn322, 2020-01-30:
Btw.. my working switch has v 1.08 while the firmware that you sent is 1.06

amitn322, 2020-01-30:
Also I now see the switch in the APP but it does not turn on.

ljalves, 2020-01-30:
If you can see it in the APP, try using the "update firmware" option in the app.
Let me know if it worked.

amitn322, 2020-01-31:
Yeah that just times out on check :(

Linearburn, 2020-02-23:
https://drive.google.com/drive/folders/1mPB06ZeS8gLTq-jwH-Yf6IBn9lred7eF?usp=sharing

there are the board pics

ljalves, 2020-02-23:
Nice. It's actually a LPT100F and not LPB100 as I thought!

Can you read the IC reference near the LPT100F? Can't see it from the pics. It looks like a port expander...

ljalves, 2020-02-23:
Ok, looking at the pics:
- I2C at GPIO23(SCL) and GPIO20(SDA)
- Buttons at:
  - GPIO18(UP)
  - GPIO11(DOWN)
  - GPIO12(ON)

I2C is used to communicate with that IC above the module (which I assume it is a 8bit PIC) which does all the work.

So, **the "only" thing missing to build a proper firmware** for this module is to know which commands to send by I2C.

There are a couple of ways to do this:
1) With a working module capture I2C traffic for each button press.
2) With a working module dump firmware, disassemble and look for the i2c code.
3) With the broken module (or a working one) read PIC flash mem and disassemble code (it may not work as PICs allow to protect firmware from being read back).

Can anyone do any of the above?
Easiest is 1) and I could easily do it if I had a working module.

**Caution:**
NEVER program or "play" with the module while connected to the AC line! These module are powered from the AC line without any kind of isolation.

For example, to capture the I2C traffic, the board should be powered exclusively with a 3.3V supply, disconnected from the AC line.

ljalves, 2020-02-23:
Another way is to find the module firmware online.
I know that the app allows to upgrade the firmware so it has to grab it somewhere!
I don't have experience decompiling android app's... anyone?

ljalves, 2020-02-23:
@amitn322 (or someone else that's willing to get this working), please flash this firmware:
https://drive.google.com/open?id=1FzZpIYAgRgwvRiUcYk8sy_8ze2MLOSeb

Then go to:
http://<ip>/state

And copy/paste the result. Should be something like:
> I2C SCAN RESULT
> found=0, addr0=0, addr1=0, addr0=0

It does an I2C scan and it should report back the address to control the PIC (if well implemented!).

If this works, it should be half way to get it done!

Linearburn, 2020-02-23:
I2C SCAN RESULT
found=113, addr0=15, addr1=16, addr0=17

Linearburn, 2020-02-23:
do you have a discord this would be a good case for a channel to discuss things

ljalves, 2020-02-23:
What the f...??
Well... not really a surprise having a crappy i2c slave implemented by them...

Makes things harder but doable I think.
I will make a web gui for you to try and send some values to those addresses to see if we can get the lights to turn on.

ljalves, 2020-02-23:
> do you have a discord this would be a good case for a channel to discuss things

Not really... I'll see what I can do.

ljalves, 2020-02-23:
> I2C SCAN RESULT
> found=113, addr0=15, addr1=16, addr0=17

When you hit refresh do you always get the same values? I hope so...

Linearburn, 2020-02-23:
2 secs

Linearburn, 2020-02-23:
I2C SCAN RESULT
found=128, addr0=0, addr1=1, addr0=2

ljalves, 2020-02-23:
Oh wait! There's a bug on the code... let me recompile it. And add more stuff to test.

Linearburn, 2020-02-23:
> Oh wait! There's a bug on the code... let me recompile it. And add more stuff to test.

np ill flash it when its up

Linearburn, 2020-02-23:
Just wondering how much spare flash space do these modules have?

ljalves, 2020-02-24:
Ok... bugs fixed. New FW to test i2c:
https://drive.google.com/open?id=1gijtTdLMlO-xfW2nngJX1OH83DXQACFM
(please paste output)

---
About flash space for webpages, it has a lot but the limitation on these modules is that a webpage is limited to 1400 bytes.
There is an alternative, which consists in rebuilding the main webpage with additional files. I can show you how to do this later.
Any help on the webpages are welcome!

Linearburn, 2020-02-24:
I2C SCAN
0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127 | t=128

Linearburn, 2020-02-24:
was that what you were looking for?

ljalves, 2020-02-24:
ok, same as before... The PIC is answering to all slave addresses... not promising.
I would suggest a hard-reboot (remove power and reapply) and test that again.
If the result is the same, then it's really a crappy i2c implementation...

To make some sense of what we are doing:
The PIC controls the output of the dimmer.
Looking at the board, I can see 5 leds so I would expect 5 levels of brightness.
0=off, 5=full brigthness
I would expect to be able to write a value somewhere to set the dimming level.

I don't know what to write to the i2c bus, but we can try guessing...
You can write to the i2c bus with:
http://_module_ip_/state?a=6&v=1     (It will write 'v'alue 1 to 'a'ddress 6)

Then try changing 'a' until something happens.

If no luck, I will change the code to write '1' or '5' to all addresses and see if we can get it to turn ON...

If I had a working module, I could easily "sniff" the correct values...

Linearburn, 2020-02-24:
ill eat dinner then ill go trip the breaker and turn it back on

[46 more replies]

closed 2020-03-11 by ljalves