ljalves/hfeasy #9: Turns itself on automatically

phrogg, 2020-01-30:
Hay thanks for the great fw!

I have the problem that the switch seems to turn itself on sometime when its off. It's strange since it only turns the thing on but never off. I monitored a bit of traffic and I couldn't find anything suspicious that sends a ON signal...

ljalves, 2020-01-30:
The problem should be current spikes in the AC line. I bet that it happens when you turn on/off something in your house (or something turns on/off automatically like a fridge).
It also happens to me from time to time but I never bothered to fix it...

How often does it happen to you?

I will improve the button debouncing code and give you a testing version.

phrogg, 2020-01-30:
Cool, thanks for clearing things up for me! I wouldn't bother to create an issue, but this seems to happen to me a lot. Like every 2 hours or so it turns itself on...

phrogg, 2020-02-02:
You are right I noticed today that it also turns itself off when its on.

ljalves, 2020-02-24:
Please flash release 0v5 and let me know if this issue got solved.
You can find it on the shared google drive folder.

phrogg, 2020-02-25:
I did something wrong while testing. I will report tomorrow how it goes, what I noticed so far was that when I try to turn it on I have to click more than 2 to 3 times quickly in a row to turn it on.

ljalves, 2020-02-25:
That doesn't make sense.
Btw, if you toggle 3 times in a row (6clicks) it will factory reset and turn into AP mode.

You are talking about the wall module, right?

phrogg, 2020-02-26:
I'm talking about the plug. When I try to turn it on via the website "/status?sw=1" it does not turn on, only when I click several times in a row. I will do some more specific and detailed testing tomorrow.

ljalves, 2020-02-27:
I thought you were talking about the wall mount module...
As far as I know, the plug doesn't have any issue and works perfectly on all my testing.
How is your setup? Are you using mqtt or just http control?

ljalves, 2020-02-27:
By the way, the next time your plug turns on (or off) "automatically", go to http://ip/status
and check "GPIO Status" - you should be able to see what caused to turn on/off.
Example:
"[...] **last changed by: HTTP**, publish: 0, action: 0"

ljalves, 2020-03-16:
No feedback for more than 2 weeks - closing.

ljalves, 2020-03-17:
@polmdm seems to have the same issue (was being discussed elsewhere).
Please continue here.

polmdm, 2020-03-17:
Hi,
yes this issue seems to be more correct.
Shall me explain exactly what happens:
on HomeAssistant I have a switch that control the Ankuoo (through MQTT), when I turn it on actually the relay closes and the light turn on but on HA the switch, after few seconds switch again in OFF position while the light remain on.
To turn it off I need to press twice the switch. In that way, I assume, the first press makes HA synced with current switch position whereas the second one send the signal to turn it off.

ljalves, 2020-03-18:
Ok, I understand it now.
Try this: monitor the subscribed topic with:
mosquitto_sub -t cmnd/abatjour/POWER

Then check there is anything going on on mqtt when HA switches OFF.
(when you turn it on, you should see the ON command, and nothing else)

About the model, you're right... they didn't specify the model in the sticker... not sure how to announce support for it on main page. The full name is "Ankuoo REC"?

polmdm, 2020-03-18:
Hi,
I've subscribed to topic and when I turn on it returns 1, when I doble press it returns 0. NO others messages...
[image]

About the model, on the switch (on front side) it's written Ankuoo REC but on the website you can find it under Ankuoo NEO: https://www.ankuoo.com/products/?sort=1

ljalves, 2020-03-18:
So, when in HA it turns off after a few seconds, you don't get anything in mosquitto_sub.

This makes me believe that something is wrong in HA setup... I don't have experience with HA so I can't help.
Could it be that the "button" you setup is a "push button" type? (automatic return to 0)?

polmdm, 2020-03-18:
Hi,
exactly, I thought being an issue linked to: the plug do not send back to HA it’s status but from the test seems to work properly. I’ll ask on some HA forums. By now thanks you very much.

PS. The switch seems to turn on after one minute when i restart it (by http), could be a timer bug?

ljalves, 2020-03-19:
> PS. The switch seems to turn on after one minute when i restart it (by http), could be a timer bug?

Use the same methods I pointed before:
After that 1 minute, go to the "http://ip/status" page and check in the gpio status what caused it to turn on (check "last changed by").
Also use "mosquitto_sub -v -t cmnd/abatjour/POWER" to monitor arriving messages to the command topic.

That should give you enough information to understand what's causing it to turn on.

polmdm, 2020-03-19:
Unplugging the switch for some minutes seems to have resolved the problem! Thanks

polmdm, 2020-03-20:
Hi,
according to what I found on HA forums (https://community.home-assistant.io/t/toggle-keep-switching-off/179699?u=polmdm), the issue seems be linked to the fact that the plug do not publish it's own state over MQTT. Maybe is the kind of plug I flashed not working properly with this firmware or there's could be a solution?

ljalves, 2020-03-20:
That's true, when I receive the cmd via mqtt I do not publish it back.

I'll change the code to publish it back if the pub topic is different that the sub.
That should fix it.

I'll let you know once I have a new fw to test.

polmdm, 2020-03-21:
WOW, that would be great!

I'll be waiting for updates, Thanks!

ljalves, 2020-03-21:
Can you try flashing FW 0v7? (available on gdrive)
Let me now if it works.

polmdm, 2020-03-21:
Okay, I've just flashed the new software but the situation is even worse: the plug is uncontrollable through mqtt, it changes the state on status page but the light doesn't turn on, and the switch stays always ON on HA

phrogg, 2020-03-21:
@polmdm yeah same issue for me with the new version.

ljalves, 2020-03-21:
Not sure what happened but there was an issue building the firmware images...
I've rebuilt and replaced them on gdrive - please re-download and try again...

ljalves, 2020-03-21:
By the way, I had some trouble to re-flash from the broken firmware - the firmware upload was always failing. Then, after some tries I'm not sure if it has anything to do with it but I disabled the LED and mqtt client (set mqtt port to 0) (submited to flash, rebooted) and I was able to recover the plug.
Or it was just a poor wifi connection...

polmdm, 2020-03-21:
Yes, it works!

(with previous version it saves the mqtt config, with this one I had to restore manually it)

Thanks you very much!!

ljalves, 2020-03-22:
@renzo56789 I think you had this issue (reported in #3).
Anyway, fixed in the latest FW.

closed 2020-03-22 by ljalves