ljalves/hfeasy #10: HF-LPB130 - Upgrade

Charnay, 2020-03-02:
Hello, First of all, many thanks for your involvement,

I doing to try Chacon socket and I found your git to manage it via URL, I think to improve security it's so much better than manage directly by the cloud application..

I encoutered the problem related that I can't upgrade my socket with your bin,
What do you need to develop the fix ?

ljalves, 2020-03-02:
Intersting... another user already emailed me about his device not working with this firmware.
Then we discovered that it uses a different module (the LPB130) so I added a warning message on the README.md about this:
https://github.com/ljalves/hfeasy#hfeasy

Is your device this one: http://www.lumitek.cn/en/productsd.php?pid=1094
Check the model on the back sticker. Is it a "SW7201"?
I'm wondering if it is just the FR version or if all the newer products use the LPB130 module.

Anyway, this firmware should work and compile just fine on the LPB130.
I could find out that there are 2 module types: one with 1Mb and other with 2Mb flash.
After discovering which one they used on the device, get the toolchain (arm-none-eabi, available for linux and windows) and the SDK from HF:
http://www.hi-flying.com/download-center-1/software-development-kit-1/download-item-hf-lpx30-hsf-sdk-1mb-2mb
Then compile and test...
If I find one of these modules I'll build the firmware.

By the way, the LPB130 is a newer version of the LPB100 and major differences are:
|          |LPB130 | LPB100 |
|-------|---------|---------|
| CPU  | ARM Cortex-M4@160MHz | ARM Cortex-M3@96MHz |
| OS    | mbed | FreeRTOS |
| Flash | 1Mb/2Mb/4Mb| 2Mb |
| RAM | 352Kb | 128Kb |

Charnay, 2020-03-02:
Hello Luis,

Yes it’s interesting ;-)

Yes I rode this, 

Yes it is, SW7201 ref53012

> After discovering which one they used on the device
Do I need unpackage the socket to find it ?
> If I find one of these modules I'll build the firmware.

How can I help you ?

ljalves, 2020-03-02:
I looked at the lpb130 SDK docs and the internal webserver doesn't seem to have the same capabilities (add custom pages on code), meaning that the fw will not work as is (mqtt should work fine, but controlling by http will not).

Since lpb100 internal webserver is also pretty limited maybe I'll just code a simple webserver (or find an opensource one online) that will work on both modules.

I already installed the gcc toolchain (has to bee the 2016-q3 version) and SDK - everything seems to compile ok so all I need it to get my hands on one of the new modules...

Charnay, 2020-03-03:
I bought my socket here https://www.amazon.fr/CHACON-53012-Chacon-Prise-Intelligente/dp/B01HX3DMYC/ref=sr_1_4?__mk_fr_FR=ÅMÅŽÕÑ&keywords=chacon&qid=1583221497&sr=8-4
yesterday 

I will donate some bucks from PayPal if you would buy this model,

Tell me what you want,

ljalves, 2020-03-03:
Not sure if that's a good idea.
It can happen that I order one and it arrives with the older module (lpb100).

Charnay, 2020-03-04:
Leave me your address at clement@charnay.cc, hope you're live not too far from france,
I will send you one of my sockets
Thanks

ljalves, 2020-03-11:
I now have a LPB130 module in my hands.
Will dump original firmware and start doing some tests.

I'll post here whenever I have a working hfeasy firmware for it.

ljalves, 2020-03-15:
An update on supporting this module:
I now have 2 of these modules in my hands: 1 is "bricked" from a wrong flash procedure, the other is working and has the original firmware in it.
I've removed the metal shield from the module expecting to see a flash rom... nothing!
This new module has everything integrated inside 1 single IC (explains why it is cheaper).

The bricked module can't even start the serial bootloader. I've connected the debug port and it spits out an hard fault error:
```
#E HardFault_Handler                                                           
[Hard Faults Report - all numbers in hex]                                       
R0 = 0x00000000                                                                 
R1 = 0x00000006                                                                 
R2 = 0x00000000                                                                 
R3 = 0x00000000                                                                 
R12 = 0x08001001                                                                
LR [R14] = 0x00001a63                                                           
PC [R15] = 0x080010cc                                                           
PSR = 0x01000000                                                                
PRIMASK = 1, IRQ:Disable                                                        
FAULTMASK = 0x00000000                                                          
BASEPRI = 0x00000000                                                            
CONTROL = 0x00000000                                                            
BFAR = 0xe000ed38                                                               
CFSR = 0x00010000                                                               
HFSR = 0x40000000                                                               
DFSR = 0x00000000                                                               
AFSR = 0x00000000                                                               
SCB_SHCSR = 0x00070000                                                          
                                                                                
                                                                                
[Detail report]                                                                 
APSR Flags: nzcv                                                                
ProcStack:0x00000000 MainStack:0x401107b0                                       
                                                                                
Memory Manage Faults                                                            
MM_FAULT_STAT:0x00                                                              
        IACCVIOL:0x00           IACCVIOL:0x00                                   
        MUNSTKERR:0x00          MSTKERR:0x00                                    
        MLSPERR:0x00            MMARVALID:0x00                                  
Bus Faults                                                                      
BUS_FAULT_STAT:0x00                                                             
        IBUSERR:0x00            PRECISERR:0x00                                  
        IMPRECISERR:0x00        UNSTKERR:0x00                                   
        STKERR:0x00             LSPERR:0x00                                     
        BFARVALID:0x00                                                          
Usage Faults                                                                    
USG_FAULT_STAT:0x01                                                             
        UNDEFINSTR:0x01         INVSTATE:0x00                                   
        INVPC:0x00              UFSR.NOCP:0x00                                  
        UNALIGNED:0x00          DIVBYZERO:0x00                                  
                                                                                
[Report done]                                                                   
Panic...           

```

Also I don't want to replace the original firmware in the working module before I dump it first.
So, to be able to do that I ordered and I'm now waiting for a new programmer that should support this cpu.

More news once the programmer arrives...

urbamax, 2020-12-23:
Hi! I got a Chacon Plug with LPB130 too... What's up about the custom firmware...?

Arut42, 2021-05-27:
@ljalves you can't brick these devices ;-)
you can always program a new firmware over Uart 0 (debug Uart)
Just get these Programm from HF => "SingleDnload_V01.00.10_20180307"

I have one of these devices On the Label HF-LPB100-0 but it looks totaly different and inside is a HF-LPB230

ljalves, 2022-01-21:
Yes, but we'll never be able to get back to the original FW (no download option)...

I'll upload the FW for the LPB130 devices soon...

Arut42, 2022-01-22:
U say u have a second one.
Just download the FW over UART0
In the Bootloader u can use the command => flashinfo
U get all info about the flash chip u need
0x18000000 - 0x18001000 is internal Bootloader u can't write to it
with => upload 0x18001000 0x200000 (<start> <len>) len 0x100000 for 1MB flash
u get a full upload of the 2MB flash witch u can download to the brick one ;)

ljalves, 2022-01-22:
Interesting... I don't remember seeing any command to get the flash contents.
I'll try again as soon as I can!

ljalves, 2022-01-25:
HFEASY 0v11 build for LPB130
[HFEASY_LPB130_UPGRADE_0V11.zip](https://github.com/ljalves/hfeasy/files/7937294/HFEASY_LPB130_UPGRADE_0V11.zip)

Let me know of any issue...

ljalves, 2022-01-26:
Official release available:
https://github.com/ljalves/hfeasy/releases/tag/0v11

Let me know how it works...

urbamax, 2022-01-26:
First, thank you for your job...!!

But... Can't access to any web page when I try "http://<deviceip>/iweb.html" or "http://<deviceip>/" ...
I'm sure about IP of the product because I gave it a static IP.
I tried with Safari and Chrome.
Chrome's telling me "ERR_CONNECTION_REFUSED"

Can you help me with the procedure to upgrade the firmware please...?

ljalves, 2022-01-27:
New release:
https://github.com/ljalves/hfeasy/releases/tag/0v12

ljalves, 2022-01-30:
LPB130 plugs are now supported. For questions/discussions post [here](https://github.com/ljalves/hfeasy/discussions).
Closing issue.

closed 2022-01-30 by ljalves