ljalves/hfeasy #11: How to get the state

Marius-Perju, 2020-03-03:
Hi, thank you for your solution, it works great with the http, I'm new to coding and I'm still trying to understand how to work with mqtt (I have openhab 2.5.2 now so the documentation is not so much and not so clear)

Would it be possible to explain how to get the actual state of the relay on an .items file? I can see the state by accessing http://ip/state but I would like the item to be updated when the state changes.

Thank you

ljalves, 2020-03-10:
I also use openhab but I use MQTT (it's much easier) - also I'm not sure how to do it with http.

Marius-Perju, 2020-03-12:
Hi Luis,

Thank you so much. Do you mind if I ask you a favor? I was reading many
topics but still I am not able to control it via MQTT.
I am running OH2.5.2, configured the device:

Server: RPi IP address
Port: 1883
Subscribe/Publish: CT/releu

I am working via files and I created the MQTT bridge and the Thing, created
a virtual number which changes from the console commands but for this
device I am not able to see it switching the state.

I tryied to configure the Items using {mqtt="broker:topic:devicename:releu"}
Also {channel="mqtt:broker:topic:CT/releu:releu"}
And also other combinations seen on the forums but nothing works.

I am using Mosquitto.

Thank you so much for your help

Marius

ljalves, 2020-03-15:
Check if the module<->mosquitto communication is working first.
You can do it by using the linux command line, in your case (if you have "1" configured as "ON value" and "0" as "OFF value"):
mosquitto_pub -t CT/releu -m "1"   -------->    should turn it on
mosquitto_pub -t CT/releu -m "0"   -------->    should turn it off
(assuming you are running this on the same machine as the mosquitto server, else you need to specify the host with -h)

Finally, work on openhab until you see the correct messages arriving to the subscribed topic.
And to do that, on a different console type this command:
mosquitto_sub -t CT/releu

Marius-Perju, 2020-03-23:
Hi,

Sorry I was away for a few days. This is what I get when I try to use those
commands:

[image: image.png]

În dum., 22 mar. 2020 la 11:23, Luis Alves <notifications@github.com> a
scris:

> Any news?
>
> <https://github.com/ljalves/hfeasy/issues/11#issuecomment-602170003>, or
> unsubscribe

-- 

Cu stima / Best regards,

-----------------------------
Marius PERJU

Automation Engineer

Address: Sinmihaiu German

307381 Intr. 87c

Timis, Romania

Mobile: +40754 528 649 <+40%20754%20528%20649>

perjumaryus@gmail.com

ljalves, 2020-03-24:
I can't see the image...

Marius-Perju, 2020-03-24:
I tryied to attach it

ljalves, 2020-03-25:
Copy/paste the text here instead of a image...

Marius-Perju, 2020-03-25:
I'm usin Putty and when I try your command the system is asking for a
password for the "openhabian", I type the password and the reply is:
"Sorry, try again."

If I try the password of the OS it sais that the password is not correct,
therefore I try the password of the mosquitto which I mentioned above.

ljalves, 2020-03-25:
If you don't know the passwords of your system of course that you can´t make it work...
Check file: /etc/openhab2/services/mqtt.cfg
It should have the password for mosquitto mqtt server.

Well, since this is not a bug/issue I'm closing it.
Regards.

closed 2020-03-25 by ljalves