ljalves/hfeasy #12: support for AW50011 non dimmer

Linearburn, 2020-03-11:
https://www.amazon.com/dp/B00N0S47NK/ref=twister_B07WPT7TTW?_encoding=UTF8&psc=1 

I'm pretty sure it's using the same controller as the dimmer module

Linearburn, 2020-03-11:
I will be ordering a new switch soon.  so i don't have to remove the one i got mounted in my kitchen lol then ill get you the pictures ect

Linearburn, 2020-04-08:
Ill buy you https://www.amazon.com/gp/product/B071FP7TW5/ref=ppx_yo_dt_b_asin_title_o02_s03?ie=UTF8&psc=1 if you can port the firmware for it

Linearburn, 2020-04-08:
tossed you 15 eur

zmesser, 2021-04-05:
I flashed your firmware to my AW50011 non dimmer and it connects to MQTT server just fine.  The only issue is that The physical push button no longer controls the internal relay.  When I send it a 1 it responds by turning on the WIFI light, 0 turns the WIFI led off.  I believe adjusting the pin mapping may be all that is needed to make it completely compatible.  Of course, I could be way off too.

ljalves, 2021-04-05:
@zmesser , you are correct. All that's needed is to fin the correct pin mapping.
Are you able to open the device and take some closeup photos of the board?
That should allow me to see what's connected to what.

zmesser, 2021-04-10:
> @zmesser , you are correct. All that's needed is to fin the correct pin mapping.
> Are you able to open the device and take some closeup photos of the board?
> That should allow me to see what's connected to what.

Of Course.  I'll post them soon.  Thank you for your work!

zmesser, 2021-04-10:
Here you are.
[image]
[image]
[image]
[image]
[image]

zmesser, 2021-05-10:
> Here you are.
> [image]
> [image]
> [image]
> [image]
> [image]

@ljalves will these images work?  I'm sure you are busy, just wondering if there is anything else I can do to help.

brmcc, 2021-09-12:
First, I love the work Luis has put into this!  I didn't realize how much until I started digging around.  I had a few on the non dimmer ones and finally had time to get the compiler up and running and was able to get them to work, mostly.

I could not get the wifi LED to work.  I tried to handle it while using mostly the plug code but ended up soft bricking my switch where I had to hook into the serial port and send a new image over with xmodem.  That was fun!

I did modify the MQTT code slightly.  For Home Assistant, I wanted a way to get the status on startup and for it not to get hung if trying to switch to the value the switch was already at.  So now, it will send back the status regardless if the switch changed or not.  Also, if you send it anything other than your defined OFF and ON commands, it will just return the current status via MQTT so you can query it's current state.

So, for what it's worth, below is a link to the one I compiled.  If you don't mind the wifi LED not working it should work fine for you, however use at your own risk, don't blame me if your switch explodes after you install it.  

For reference, here are the pins used:
Green Wifi LED - 10
Power LED - 19
Relay -18
Button - 12

[http://www.shcircuit.com/HFEASY_AW51011_UPGRADE.bin](url)

ljalves, 2022-01-22:
Nice work.
I finally have some time to work on this again and would be great to add your enhancements.
I'll try to figure out the led issue.

ljalves, 2022-01-22:
That pin mapping you posted doesn't seem right.
The LPT100F module pins are mapped to the sdk as follows:

| module pin | function | sdk_gpio_nr |
|-|-|-|
| 1 | gnd | 1 |
| 2 | 3V3 | 9 |
| 3 | nReload | 45 |
| 4 | reset | 47 |
| 5 | uart0_rx | 41 | 
| 6 | uart0_tx | 39 |
| 7 | pwr_sw | 25 |
| 8 | gpio | 18 |
| 9 | gpio | 12 |
| 10 | gpio | 11 |

In your post this translates:
relay = module pin 8 (gpio18)
button = module pin 9 (gpio12)
power led = 19? (isn't this led directly connected to 3V3?)
WiFi led = 10? (could it be module pin 10 which translates to gpio11?)

brmcc, 2022-01-22:
I'm glad you're looking at adding the switch.  I've had it working great for a few months now except for that green wifi LED.  I did not know where to really put it in the code, but I thought I could toggle it on and off having 10 defined in the relay position.  From seeing the mapping above, that makes no sense.

These do work, for sure, though.

	HF_M_PIN(19),	// led
	HF_M_PIN(18),	// relay
	HF_M_PIN(12),	// switch/button

At least, the button does turn on/off the relay as expected and the orange power light lights up as it should.  I really wasn't sure where to put the WiFi LED though.  It will go into the "recovery" mode though if you hold the button in, just like the dimmers do.

brmcc, 2022-01-22:
I'm going to recompile with pin 9 instead of 19 for the LED and see what that does..

brmcc, 2022-01-22:
Ok, my apologies, 9 as well as, for whatever reason, 19 controls the power LED.

brmcc, 2022-01-22:
Ok, I was assuming it was the power LED that used the LED field and not the WiFi led function.
Now the WiFI LED works as it should, as well as the power LED.  Here's the final values:

	HF_M_PIN(11),	// led
	HF_M_PIN(18),	// relay
	HF_M_PIN(12),	// switch/button

Thanks for figuring that out!  With those values everything works perfectly.

ljalves, 2022-01-22:
That makes sense! :)
I'm actually changing the code to allow gpio config on a web config.
That should make it pretty flexible in adding new modules.

brmcc, 2022-01-22:
That will be awesome!  I really didn't know what I was doing, just started from one and worked my way up until it did something.

ljalves, 2022-01-22:
Whenever you can please test latest release (https://github.com/ljalves/hfeasy/releases/tag/0v9).
Select the "us wall switch" on the device menu, save to flash and restart.
Let me know if all works. Thanks!

brmcc, 2022-01-23:
The switch works great.  Even the WiFi LED changes according to the setting!!  :)  I think you can officially call this one supported!

ljalves, 2022-01-23:
Thanks for testing

closed 2022-01-23 by ljalves