ljalves/hfeasy #13: Unable to flash US Wall Dimmer (AW51011D) - HF-LPB100

chaudhryfaisal, 2020-04-07:
Hello, First of all, many thanks for your involvement and development,

I am trying to update AW51011D and it seems to have no impact. I tried to upload *HFEASY0v7_DIMMER_UPGRADE.bin* and after upgrade goes through it present same web page again as if nothing changes. I also tried *ORIGINAL_FW_DIMMER_1_5115.BIN.bin* and it again had no effect on it. 

[image]
[image]

ljalves, 2020-04-07:
Did you read the README.md? Main webpage does not change!
After flashing hfeasy just check one of its webpages works (like http://ip/status).

chaudhryfaisal, 2020-04-07:
yes I thought I updated this post but it was in draft 
I was able to flash it with your firmware. my next question was how to control brightness but I figured it out by invoking state?sw=1-128

thank you again for the amazing work you have done.

closed 2020-04-08 by chaudhryfaisal