ljalves/hfeasy #14: HF-LPT100F bricked

kikkopikk, 2020-04-12:
Hi,
i have upload the HFEASY0v7_MODULE_UPGRADE.bin on my Anuoo rec.
After that, it worked, for some sec and after that it is offline. I have tried to "factory reset" it (6/10 turn off/on in 3sec) and it only turn on the light with a flash and nothing else. 
I tried to remove the 220v, but no wifi segnal or acustic sound.
How can i repare it? 
[image]

Linearburn, 2020-04-12:
does it do anything?

kikkopikk, 2020-04-13:
Nope,
I works only as physical rely, not smart.
It's offline and i can't reach It by its wifi network ip.

ljalves, 2020-04-14:
Did you check if it is in AP mode?
Scan for open wireless networks (usually named REC....).

kikkopikk, 2020-04-14:
I see the network HF-LPB100, and now i reach the rely. and it's Solved
but...
I set again the wifi home network, and it works good with http call.
But i cannot add it to the rec app on my phone and so i can't link it with
alexa.
How can i do that?

ljalves, 2020-04-15:
This firmware doesn't work with the original REC app - the whole goal is to have the devices not connected to the whole world.
It's meant to be used with either http or mqtt control.

You can use a couple of home automation software to make it work with Alexa (like openhab2, etc).
Just google for a couple of guides.

Closing issue.

closed 2020-04-15 by ljalves