ljalves/hfeasy #15: Mqtt Client formating

Linearburn, 2020-04-15:
Thanks for your hard work first of all. 

This may be more of I'm confused topic but here it goes. 

I'm trying to make sense of how hfeasy handles MQTT. Using MQTT explorer I see the topic it creates but it's just a single value and I can't adjust the brightness via MQTT It's either on or off. Im fimular with how tasmota handles MQTT and its fairly easy to configure. but hfeasy it not laid out anything like I would expect and the messages it publishes are super simple. Is there any way to add auto-discovery a full topic entry. I have attached a screen shot of what i mean.  [http://image.xvxagaming.com/images/2020/04/15/mqtas.png](url)

Linearburn, 2020-04-25:
So after some exploring MQTT, it seems it is only partially supported. While it does work to brighten and dim  Seems lot of MQTT options are missing. Below is the current output of the dimmer between 0-128 I pulled values from a martin jerry dimmer for comparison.

Current Output
REC/HA-Back-Porch-Living-Room/
=128

Expected Output 
REC/HA-Back-Porch-Living-Room/
RESULT = {"POWER":"ON"}
POWER=ON
BUTTON1T = {"TRIG":"TOGGLE"}
BRIGHTNESS = {128}

Other features not currently supported "Last will and testament" "Birth Message" "HeartBeat??"

ljalves, 2022-01-22:
I'm not very experienced with mqtt but with some help we can improve it.
Latest release should be better (last will added).
Let me know what else should I change.

ljalves, 2022-01-28:
Version [v1.0](https://github.com/ljalves/hfeasy/releases/tag/1v0) has improved MQTT messaging (LWT, autodiscovery for homeassistant).
I'll close this issue but if you have more improvement suggestions re-open it (or a new one).

closed 2022-01-28 by ljalves