ljalves/hfeasy #17: MQTT wont connect

mach7, 2020-05-26:
I've placed my mqtt information into the config page and double checked everything, all my information is correct. Same information that I use to connect other mqtt clients. But for some reason I can not get the /status page to change to connected. I'm trying to dig into this further by looking for some sort of a error log so I know what it is doing after I flash and restart for mqtt. Is there such a log?

Linearburn, 2020-06-17:
it may be a bug Mine does connect it just don't reflect it in the webpage

brmcc, 2020-06-21:
I just started using this and never could get MQTT to connect either.  There was no network traffic at all to the MQTT server.  I finally remembered I had it in static IP mode and went back to DHCP.  It worked fine in DHCP mode, but will not work at all if you have a static IP on the switch.

ljalves, 2022-01-21:
If the device is setup with a static IP, MQTT doesn't connect??

brmcc, 2022-01-22:
That is correct.  I never could get any of the plugs/switches to work in static IP mode.  I could access the switch directly from the MQTT server so it did have good connectivity, and the web page and so forth worked.  I just ended up assigning an IP from my router but it was really confusing at first.  Thanks for publishing this awesome software though, it works great otherwise!

ljalves, 2022-01-22:
Please try with latest fw: https://github.com/ljalves/hfeasy/releases/tag/0v8
I think the issue was related to the DNS server (that you were not setting when using static ip, right?)

Anyway, it should work now.

ljalves, 2022-01-22:
Whenever you can, please confirm if this is fixed on the latest release.

brmcc, 2022-01-23:
Thanks for working on this and especially a huge thanks for adding LWT.  I had kind of hacked that in to my builds, it's a very nice feature!
Unfortunately I still can't get it to connect to the MQTT server. I did actually have a DNS server in there as it required one.  I can go to the web pages directly from the MQTT server so it should be able to connect OK.  I just don't see any MQTT traffic at all.  I'm confused why it would not work but the HTTP does.

brmcc, 2022-01-23:
Just thought I would add the MQTT light just flashes in a slow on/off pattern.

brmcc, 2022-01-23:
Just an FYI with the new release.  I put the wall switch back on DHCP and was just changing the MQTT topics to get it working better.  When I hit apply, it took it a very long time, then just returned 404 errors, even on iweb.html.  I power cycled it but it seems totally dead now.  Fortunately, I do already have serial port hookups soldered in so can recover it easier.

brmcc, 2022-01-23:
My apologies for writing too soon, I was able to get it back using power button to go in recovery mode.   It's all fine again now! :) :) :)

ljalves, 2022-01-23:
Test fix with:
0v10 'beta' release: [HFEASY_UPGRADE.zip](https://github.com/ljalves/hfeasy/files/7920128/HFEASY_UPGRADE.zip)

closed 2022-01-23 by ljalves

brmcc, 2022-01-23:
It's working great with the release 0v10!  Thanks for fixing that!!!  :)

commit 0b7811fa (2022-01-23): config: set has_ip when ip is static

Should fix #17