ljalves/hfeasy #20: Support for RGB Controllers (Feature request)

13mralex, 2020-10-23:
I own a few of [these](https://www.amazon.com/XCSOURCE-Channels-Controller-Android-LD686/dp/B01AA6221S/ref=sr_1_8?dchild=1&keywords=xcsource+rgb&qid=1603415329&sr=8-8) generic RGB LED strip controllers, little did I know when purchasing that not all of the other similar branded ones of this product don't all use the same chipsets. Other brands ([LEDENET](https://www.amazon.com/LEDENET-Smart-Controller-Channels-Control/dp/B01DY56N8U/ref=sr_1_1?dchild=1&keywords=ledenet&qid=1603415676&sr=8-1)) use an ESP. The one I bought of course doesn't have that, but does have this chipset (LPB100-1).
I've lived with the built-in HomeAssistant integration, but reliability and responsiveness is usually 50/50.

ljalves, 2022-01-21:
Are you able to open up the module and post some photos?
If we can find out the pin mapping than it should be easy to add support for it.

ljalves, 2022-02-01:
No answer... closing

closed 2022-02-01 by ljalves