ljalves/hfeasy #22: Plug sometimes immediately switches back to previous state

DirkBaumeister, 2020-11-22:
First of all: Thank you for your amazing work! It's great to have this kind of device to be controllable by home assistant!

Unfortunately I'm experiencing a strange problem

When I control the plug by MQTT or HTTP sometimes it switches immediately back to its previous state. For example:
If I send a 1 and the current state is 0 then it switches to 1 and immediately back to 0. And vice versa.
Strange enough this only happens sometimes.
I learned from some other issue on here that I can see the last trigger of status change on the /status page. So I did und when this problem occurs it reads the following:
`Relay: Open(Off), last changed by: SWITCH, publish: 1, action: 2`
So it seems to me that the plug magically overrides the input sometimes.

Any ideas on this one?

But thanks anyway for your project, I really like it! :)

Thank you in advance.

Zoopa, 2021-06-22:
I've got the same problem. From my observations, it seems to matter what is connected to the plug (or where in the house the plug is plugged in?).

I've been using 3 plugs for my aquarium (lights, etc) for several months, controlled by MQTT with Home Assistant. During this timeframe, I've seen the following:
- The first plug had this problem once or twice a week
- The second hat this poblem twice in 2 months
- The third plug never had the problem

I then disconnected the first plug and from the aquarium tech and plugged it into another wall outlet in my house, with nothing plugged into the plug. It is still connected to WiFi and Home Assistant,uses the same Home Assistant schedule, and still switches on and off 4 times a day. But the problem of immediately switching back to the old state has never happened since, according to the Home Assistant logs. Therefore, I assume it has something to do with whats plugged into it or which outlet is being used (or maybe whether or not there is an extension cord involved?)

ljalves, 2022-01-21:
I suspect that this is related to spikes in the AC line caused when the relay toggles.
These devices use cheap power supply circuits (sometimes transformer-less) and spikes in the AC can trigger a gpio as if it was a switch press.

I'll increase/improve the button debounce time (maybe even make it configurable).

ljalves, 2022-01-24:
I will add config settings for the button debounce timers.
That way people can increase if they have too much noise in the AC line.

DirkBaumeister, 2022-01-24:
Thank you! I am looking forward to this and I will test it. :)

Zoopa, 2022-03-02:
Great to see you working on this firmware again and thanks a lot!

Some days ago, the official app of my switches stopped working (Connection timeout). All alternative apps (Rittig, NEO, Rec Plus) in the Google Play store stopped working at the same time. Seems like the backend cloud servers are not available anymore. This might be a temporary issue, but if it's persistent, then your firmware could be the saviour for thousands of switches ;)

ljalves, 2022-04-08:
Please test with latest release (1v1).
There is now a setting under gpio config to increase the debounce timer.

ljalves, 2022-04-18:
Re-open if that doesn't solve the issue.

closed 2022-04-18 by ljalves

gefi76, 2022-09-18:
First of all i want to thank you for your great work as this code makes a wifi extension cable (OBI Steckdosenleiste 3-fach Weiß mit WIFI Alexa-kompatibel) useable for me. But i have the same error that it sometimes switches back to the off-state if i want to switch it on. I tried to set debounce-time to various values, but still no difference. 
Workaround for me is a shellscript on my raspi to turn on / off and check the state afterwards. The only problem is that it might  be bad to some devices pluged in if the get turned off again after half a second. 
If someone needs my oneliner to turn on:
#!/bin/bash
while [ $(lynx -dump http://10.0.0.22:8080/state | cut -f 8 -d \") -eq 0 ] ; do lynx -dump http://10.0.0.22:8080/state?sw=1 ;done

Maybe you find a solution in the firmware for this issue.