ljalves/hfeasy #23: HF-LB100 in Lidl Silvercrest Wifi Plug

LubosKovar, 2020-12-30:
Thank you for great work!

I have Wifi Plug by Lidl brand Silvercrest. There is HF-LPB100 and I hope for upgrade by your firmware and connect it to the homebridge. Unfortunately I can not flash it. I did it by web browser - no sucess. I try special link as your advice: (http://<deviceip>/iweb.html)) - no sucess. After uloading browser gave me Error message: server IP_Adress no response. Firmware still the same.

Is your code working with this type of WiFi Plug even if HF-LPB100 is inside? Or does I do something wrong?

May thans for help. 

Status screen from browser:
[image]

peter--s, 2021-05-17:
Hello, 

Stumbled upon this project and this issue while investigating the contents and potential of my new [REV WiFi plug](https://www.rev.de/DE_produkt_13876.ahtml).
While i have not tried to flash hfeasy yet and I am still unfamiliar with the HF-LPT100F module, I am very surprised that your attempt was not successful because both modules seem to share both the SDK and the MCU.
I found some information that you might find useful while waiting for or finding an answer (some might be off-topic, though, since not specific to Lidl plugs):
- [openHAB related, but very detailed](https://community.openhab.org/t/solved-g-homa-wifi-smart-socket-emw302wf-and-emw302wfo-with-openhab2/21524)
- [FHEM related, with h/w details](https://wiki.fhem.de/wiki/G-Homa) and [matching forum thread](https://forum.fhem.de/index.php/topic,41362.0.html)
- [manufacturer homepage small DIP modules overview](http://www.hi-flying.com/embedded-wi-fi-module-series/low-power-module), specific [hf-lpt100f details, including downloads, e.g. firmware 1.0.08](http://www.hi-flying.com/hf-lpt100f) and [SDK](http://www.hi-flying.com/download-center-1/software-development-kit-1/download-item-lpb100-lpt100-lpt200-lpb100u-lpb105-sdk)
- [WiFi to serial application for arduino](http://2xod.com/articles/Arduino_Wifi_With_Hi_Flying_HF-LPT100_or_USR-WIFI232/)
- [WiFi to serial application](http://www.rei-labs.net/usr-wifi232-wifi-to-serial-module-hf-lpt100/)
- [HF-LPB100 serial firmware download](https://docs.gizwits.com/en-us/DeviceDev/Debug/HF-LPB100.html)

Good luck!
Peter

ljalves, 2022-01-22:
@LubosKovar,
What browser did you use? I found out that firefox doesn't always work.
Try Chrome and use latest firmware: https://github.com/ljalves/hfeasy/releases/tag/0v9
You will need to open the device and find out which pins connect to the relay/button/led.
I can help if you post some photos.

Sturi2011, 2022-01-23:
Hi,

rev Wifi Plug is the same as GHOMA OBI plug.

Silvercrest is similar but seems to have changed pinout und multicolor LED

https://www.epanorama.net/blog/2016/07/18/silvercrest-sws-a1-iot-power-outlet/

Gretings Andreas

ljalves, 2022-01-23:
Did anyone try the latest firmware and found out the pin mapping?
Also, looking at the photos from the last link there is a tiny board I can't identify - it has a crystal, could it be a receiver for a remote ctrl?

Sturi2011, 2022-01-23:
Hi,

own none of the silvercrest plugs but it might by a Revier Freunde remotes or a sender for 433MHz child plugs.

peter--s, 2022-01-24:
Hi, yep, seems to be a RF transmitter based on comments regarding similar products in [this forum thread](https://forum.fhem.de/index.php/topic,38112.225.html) while somebody claims [the pinning to be the following](https://forum.fhem.de/index.php/topic,38112.msg384120.html#msg384120).

ljalves, 2022-01-28:
@LubosKovar Try uploading latest firmware (1v0) (upgrade application, not web page) and don't use Firefox.
I would love to hear from one of these plug owners if it works.

LubosKovar, 2022-01-28:
@ljalves Which upgrade application do you mean? I try it by Chrome without success.

ljalves, 2022-01-28:
No longer using gdrive. Releases are here on github now:
https://github.com/ljalves/hfeasy/releases/tag/1v0
(use binary HFEASY_WEB_UPGRADE.bin for lpb100)

The update is done using the device webserver.
Browse to: http://\<device_ip\>/iweb.html

Select file on "upgrade application" and hit "upload".

LubosKovar, 2022-01-29:
I did it. Plug is "bricked":-( after this procedure. May I ubrick it back? Or some reset procedure? The chipset inside is LB100-1

ljalves, 2022-01-29:
Are you sure? After flashing the device will enter AP mode.
Did you check for a wifi network (usually called "LPB100")?

LubosKovar, 2022-01-29:
Yes, I checked. Did not new wifi (AP) appear. LED light has constantly red color and plug does not respond to holding button to enter to any other mode...
[image]
.

ljalves, 2022-01-30:
The non-working led/buttons are normal because the fw initial state is all gpio disabled.
But it also starts the AP mode so that it can be configured...

Can you open it and solder some wires to the UART port?

ljalves, 2022-02-01:
Waiting for the plug to arrive! 
I'll give more news once it gets delivered.

ljalves, 2022-04-08:
Thanks for sending the plugs - they are now fully supported in fw release 1v1.

closed 2022-04-08 by ljalves

Snayler, 2023-01-29:
Hello, sorry for digging up a closed thread, but I'm having issues with the same plug (SWS-A1).

I was able to upload the hfeasy firmware, and got the configuration webpage, but kept getting issues with MQTT, it seems the plug spammed ON and OFF commands making it keep changing the state of the relay. I then decided to restore the configuration but it somehow bricked the device. Now the LED only lights a faint red light and there's no AP for me to connect to. Trying to force it to recovery mode doesn't work (press button 6 times within 3 seconds, or pressing for 3 seconds or more).

I have a UART adapter (ch341-uart), but I'd need help figuring out where to solder the cables. I'm guessing the UART port is that small rectangle (labeled JP3) in the photo below:
[image]

Is that correct?

LubosKovar, 2023-01-30:
Hello Miguel,

 

I registered MQTT problem too and ON/OFF problem. I have problem with internal CRONTAB ON/OFF definitions too. I looks like these commands does not reflected internat ON/OFF script.

 

Yes, it looks like jumper JP3 is for UART flashing. Unfortunately I do not know, which PIN is RX/TX/GND/3,3V

 

What about UART sold directly to PINs on CHIP? I found web: http://www.hi-flying.com/hf-lpb100

 

Best regards

 

Lubos

 

 <https://user-images.githubusercontent.com/11491485/215360692-17f06f00-e4f6-4532-af73-fba78d2e6ac8.jpg>

mercs759, 2023-04-21:
Hello, 
same "perpetual faint red" bricking after an attempt to save a setting and restart.

fully described here : https://github.com/ljalves/hfeasy/discussions/28