ljalves/hfeasy #24: Way to reconnect to apps

manupuntocl, 2021-08-02:
Thanks for your excellent work here!
Is there a way (with the new firmware) to reconnect the device with the Rec Switch/Ankuo app?
I don't have a HomeBridge device (and I can't either), in addition to the fact that my family already installed the application, and they can control the plug outside the home (via 4G, etc)
Personally I use the http request to control it more quickly, but it works only on Wifi (obviously, local ip, but works very fine and fast with iOS Shortcuts)
Thanks for your help, and excuse my English, I speak Spanish and it costs a little bit! Greetings from Chile!

jayPare, 2022-01-20:
Looking for the same. After flashing the device with new firmware, REC Plus app doesn't find the device anymore (appears offline). If you reflash the original firmware, app starts to find it again.

ljalves, 2022-01-21:
Not possible.
The whole idea behind the project is to remove the device from the cloud and make it local.

If you only have 1 device then I agree that makes no sense to install "Home Assistant" or "OpenHAB".

In that case just go into your router port-forward and setup a port for the device.
Example:

outside port: 10000 -> device_ip:80

And then control it with:

http://\<your_public_ip\>:10000/ctrl