ljalves/hfeasy #25: Ankuoo LPB130-1 based plug

livius147, 2021-09-05:
Dear all,
I am working on a ankuoo plug device based on LPB130. I am currently trying to work on:
1) understand the schematic of PCB board;
2) porting of hfeasy firmware to the new MCU.
About point 1, I see more GPIO connections than I expect. I see the following soldered pins:
1, 9, 19-Vcc
2-NC
5-GPIO5 (I don't understand where it is used)
17, 18-GND
25-UARTtx
27-UARTrx
28-GPIO22 (connected to the button)
29-GPIO8 (connected to LED2)
30-GPIO24 (used by the relay, it is connected to the base of transistor driving the relay)
31-GPIO25 (I don't understand where it is used)
This is what I found with visual/electrical inspection.

About point 2 I am trying to understand how the pins/gpios are mapped inside the hfeasy source code. In gpio.c I find the array hf_gpio_fid_to_pid_map_table: where is that array used? I don't understand well what is the use of this variable?

Please, if you have any suggestion... In particular about the point 2.
Regards.
Livius
[image]

ljalves, 2022-01-21:
I already have a working firmware for the plug based on a LPB130.
I never got to upload it because there is no way to revert to original FW.

Well... I'll upload it as soon as I can.

ljalves, 2022-01-25:
HFEASY 0v11 build for LPB130
[HFEASY_LPB130_UPGRADE_0V11.zip](https://github.com/ljalves/hfeasy/files/7937294/HFEASY_LPB130_UPGRADE_0V11.zip)

Let me know of any issue...

ljalves, 2022-01-26:
Official release available:
https://github.com/ljalves/hfeasy/releases/tag/0v11

Let me know how it works...

livius147, 2022-01-27:
Hi,
I have successfully uploaded the new firmware, I configured the wifi connection and I saved the values. After software restart I am unable to access the device. Please may you suggest me something? I would like to debug the device via uart connection: I have already soldered wires to Vcc GND TX and RX, but the uart console seems to be empty.

livius147, 2022-01-27:
I used the UART0 to check the serial connection. Maybe I used a wrong baudrate: I read in the documentation that the right speed should be 230400.

ljalves, 2022-01-27:
Seria baud is 921600

ljalves, 2022-01-27:
Before reflashing, enter the serial bootloader and select someting like "erase settings" which I believe is 'B' and press enter.
Then disconnect nReload and check if the device entered AP mode.

ljalves, 2022-01-27:
I just realized that after the conversion factory->hfeasy, the gpio will not be configured so it will not be possible to enter AP mode...
I'll update the code to fix this (always enter AP mode after initial flash).

ljalves, 2022-01-27:
[HFEASY_SERIAL_LPB130.zip](https://github.com/ljalves/hfeasy/files/7951662/HFEASY_SERIAL_LPB130.zip)

This FW has a couple more fixes and should enable AP mode after reboot.

ljalves, 2022-01-27:
@livius147 one question: when you configured the wifi connection after flashing the firmware, did you select WPA2/AES ?

livius147, 2022-01-27:
Yes I did.
Please may you give me the complete procedure to reflash via UART the device. The nReload button shows a strange behaviour.

ljalves, 2022-01-28:
> Yes I did.
That was the reason you lost access (AES was broken...)

> Please may you give me the complete procedure to reflash via UART the device. The nReload button shows a strange behaviour.
Follow this: https://github.com/ljalves/hfeasy/issues/10#issuecomment-1022241217

Also use latest release: https://github.com/ljalves/hfeasy/releases/tag/1v0

livius147, 2022-01-28:
GREAT! It is working now.
But the blue led is always switched off.

livius147, 2022-01-28:
I changed the standard config for plug device to custom: the right pin for wifi led is 31 NOT the 29.

livius147, 2022-01-28:
What is HF in the web application? I receive the following message:
ERROR:404 Not Found

livius147, 2022-01-28:
Another issue:
[image]
As you can see, with DHCP (automatic IP), the default DNS is wrong. In my case it should be equal to the gateway IP address.
In other words: instead of 10.10.100.254 it should be 192.168.0.254.
Regards.

ljalves, 2022-01-28:
Nice that it works!

'plug' is for the ankuoo plug, not all plugs. I really need to update the names.
How many leds does this plug have? From your comments, it seems 2 - are both working?
I'll create a preset config for this device - can you post a screenshot of the gpio config page?

'HF' is for the factory webpage.
If you get a "not found" you'll need to flash it again (not that it's that useful):
Go to the update page and choose "update customized webpage" and flash with:
[lpx30_webpage_2.0.1_en.bin.zip](https://github.com/ljalves/hfeasy/files/7961311/lpx30_webpage_2.0.1_en.bin.zip)
(THIS FILE IS FOR LPB130 ONLY)

Regarding the DNS, that's kind of a "bug". It's just displaying the setting in case you select "manual IP config".
The real DNS in use should be the correct one.
I'll see if I can fix that (at least hide it when using dhcp to avoid confusion).

ljalves, 2022-01-28:
Wait... this is an "Ankuoo" plug, right?
Mine has the LED at 29, are you sure it's 31?
If you select LED function "Find" and hit "Apply" does it start blinking in a fixed pattern?

livius147, 2022-01-28:
I have two leds: 
-the first one is yellow (working without changes) and it is switched on/off when I switch on/off the relay
-the second one is blue and it has the WiFI icon; this seemed not to work until I modified the gpio as previously mentioned.
[image]
As you can see the wifi led (the blue one) gpio has been modified to 31.

livius147, 2022-01-28:
Sorry, I haven't understood what you are saying about LED find function and apply... please, explain what you mean.

livius147, 2022-01-28:
[image]
I have understood: yes, when I select Find and then apply, the wifi is flashing with the default plug configuration (where the wifi led is set to 29).

livius147, 2022-01-28:
Maybe it would be better to be able to set the wifi len on when the device is connected to the network, not just when something is transmitted via HTTP.
What about the option Enable CORS* ?
Thanks a lot.

ljalves, 2022-01-28:
Sure I can add that option to the LED (personally, I don't like it always on - too much light at night! :)

CORS was a request in #18 
Basically, if you enabling that option, it allow to be controlled by other webpages using http requests.

livius147, 2022-01-29:
Hi,
I have still two questions:
1) is it possible to identify the flash size from the written code LPB130-1 in the external box, 1 stand for 1 Mega?
2) I would like to migrate your project to linux (in ordfer to compile all under linux), have you any suggestion about that? May I start from the LPB130 sdk examples? If I will succeed, if you agree, it would be useful to put Makefile or CMakeLists for linux inside the project.
Regards.

ljalves, 2022-01-29:
Hi,
1. Not sure, I can confirm but I think you are correct.
2. I already have Makefiles working for the project. I'll submit them shortly (never did because I'm forced to use windows to build LPx100 firmware...).

ljalves, 2022-01-29:
Just added Makefiles.
You'll need to download the arm gcc toolchain release 2016q3 (newer ones don't work with the provided sdk libs):
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads#panel14a

Uncompress to some folder and setup it's location at the "Makefile.mk"

ljalves, 2022-01-29:
Actually latest working arm gcc toolchain is release 7-2018-q2-update

livius147, 2022-01-30:
Thanks a lot. All is working fine.
I will try to port your makefiles to cmake.
Where did you find the tools (gen_upgrade_bin and imgpkt)? Are they in the SDK? I haven't found them.
Regards.

livius147, 2022-01-30:
Is it right that in the main Makefile there is 
CFLAGS += -D__LPT230__ 
?
Is it right that define? __LPT230__?

ljalves, 2022-01-30:
> Where did you find the tools (gen_upgrade_bin and imgpkt)? Are they in the SDK? I haven't found them.

Yes, under the "tools" folder. The only difference is that "gen_upgrade_bin" is called "LPT100_IMG_Tools".

> Is it right that define? LPT230?

Actually that define is not even needed (you can remove it).

ljalves, 2022-01-30:
By the way, why do you need CMake?

livius147, 2022-01-30:
Because I prefer that compilation system: it is useful to check all requirements (packages and libraries) and the cmake configuration can be easily extended to other OS or environments. A preference of mine, it is not a must.

ljalves, 2022-01-30:
Closing issue since this is solved / supported.

I have enabled [Discussions](https://github.com/ljalves/hfeasy/discussions) so if you have more questions please post there.
Thanks

closed 2022-01-30 by ljalves