ljalves/hfeasy #26: Config Change for GHoma Plugs

Sturi2011, 2021-11-28:
Hi, 

thans for the work. For G-Homa Style Wifi Sockets the GPIOs are changed.

The connections are the following: 

18 key
12 relay
11 led

Please provide a firmware bin for these devices as they are widley used in FHEM community
and imho mqtt is a big benefit.

Thanks

Andreas

ljalves, 2022-01-21:
I'm resuming the work on this project and on the next firmware I will try to provide a config page for the pins.

Sturi2011, 2022-01-22:
Hi,

this would be great.

ljalves, 2022-01-22:
New firmware: https://github.com/ljalves/hfeasy/releases/tag/0v8

Please try it by selecting device "custom" and setting up module type & gpios.
(save to flash and reboot at the end)

If all works, I can then add the device to the pre-configured list

PS: gpio config nrs are related to the module pin numbers (lpb100 = 1 to 48; lpt100f = 1 to 10)

ljalves, 2022-01-22:
By reading your post, you should be setting up gpios as follows:

wifi led: gpio 10
relay: gpio 9
button (push): gpio 8

Sturi2011, 2022-01-22:
After flashing it shows v0.7 and there is no device button.

 

Flashed as new firmware over iweb….

ljalves, 2022-01-22:
If it shows 0.7 then it didn't flash properly... try again please.

ljalves, 2022-01-22:
What browser are you using to flash the device?

ljalves, 2022-01-22:
For some reason I have issue when using firefox to flash the devices.
Google chrome flash it fine...

Sturi2011, 2022-01-22:
Firefox - first then Chrome - wich was then working.
Im on testting with ports.

ljalves, 2022-01-22:
Can you post a screenshot of the gpio config page?

The latest release now allows you to set thew led to match the relay state.

Sturi2011, 2022-01-22:
V0.9 is nearby perfect. hanks for your work.

A little issue - the led control is inverted (ground driven ;) )

Picture of config:
[image]

Picture of device:
[image]

Greetings

Andreas

ljalves, 2022-01-22:
Thanks for he photos. I wasn't aware that it used the lpb100 module.
About the led inverted signal, that's a good point!
I'll add a switch to invert led output (and even relay).

Sturi2011, 2022-01-22:
Best way will be a checkbox on config page IMHO

Greeting Andreas

Sturi2011, 2022-01-24:
Hi,

in config.h

{ 11, 12, 18, 0, 0, 0, 0, 0, 0, HFM_TYPE_LPB100, 0 }

should be
{ 11, 12, 18, 0, 0, 0, 0, 0, 0, HFM_TYPE_LPB100, inv}

ljalves, 2022-01-24:
I've added your device to the preset list in the latest fw:
https://github.com/ljalves/hfeasy/releases/tag/0v10
Let me know if all works ok.

I'll see what I can do about the config strings - should not be hard.

Sturi2011, 2022-01-24:
Preconfig for ghoma woks great.
MQTT Responses are as expected.
Perfect - thank you for all the work.

ljalves, 2022-01-26:
New release - MQTT LWT fixed.
https://github.com/ljalves/hfeasy/releases/tag/0v11

closed 2022-01-26 by ljalves