ljalves/hfeasy #27: hfeasy on rec plug startup mode

papampi, 2022-01-17:
I just installed hfeasy firmware on rec plug but it boots with state 0 disconnected mode.
Is there any way to make it boot with state 1, connected mode?

Also tried to restore to factory and now the app not work. 
Any ideas?

papampi, 2022-01-17:
Update:
Found out that if use on timer it will wait the time and turn on the device at reboot.

jayPare, 2022-01-20:
@papampi do you mean Timer or Countdown?

papampi, 2022-01-20:
Timer has two modes on and off each one waits the time you set for the other one to kicks in and activate, so if you set timer to "on" with 5 seconds whenever switch off it gets back on in 5 seconds also if lost power or for whatever reason unplug it waits 5 seconds and turn on the switch.

jayPare, 2022-01-20:
@papampi yeah so you meant Countdown. Timer feature is not yet implemented.

papampi, 2022-01-20:
No I mean timer
I use it with http://ip/timer

ljalves, 2022-01-21:
Let me try to provide at the config page a field to specify the power on state.

ljalves, 2022-01-22:
Feature available on release 0v9 (https://github.com/ljalves/hfeasy/releases/tag/0v9)

closed 2022-01-24 by ljalves