ljalves/hfeasy #30: Missing Countdown menu

papampi, 2022-03-04:
There used to be a Timer/Countdown menu
I see the function still available at http://<ip>/timer 
But no menu available.
Countdown is a real good feature, Please dont remove it.
And if possible add it to menus again.

Thanks

ljalves, 2022-04-08:
Timer config page added to main page.
Also added "crontab" alike functionality.

ljalves, 2022-04-08:
Available on latest release (1v1)

closed 2022-04-08 by ljalves

jef-hendrickx, 2022-11-07:
Hello I'm new on this site and used your firmware for the first time yesterday.
Its working well but I have some minor issues.

When applying new data in crontab the next action time is calculated on UTC and the value for UTC+ ( or UTC-) in the NTP client settings is not added.

My router is using spaces in the Wifi SSID.  The app accepts the spaces and the following characters; it works well when applying the  SSID.  When going back to the wifi page for changing other parameters, the characters after the spaces aren't showed .  When applying the new paramters the SSID isn't (re)loaded correctly.