ljalves/hfeasy #35: /config and /state sending "ERROR:404 Not Found"

Lodhar, 2022-05-02:
Hi everyone,

So, I have two HF-LPB100 (Chacon). The upload page is showing (iweb.html) but not the rest. I used that last page to update HFEASY_WEB_UPGRADE and then HFEASY_SERIAL_UPGRADE. But the url context /config, /state and /timer, all return a "ERROR:404 Not Found" exception. Did I do something wrong?

Some more details:
[image]

maxiido, 2022-07-28:
Hi,
as far as I know the HFEASY_SERIAL_UPGRADE is only for upgrading via a serial wired connection. So I guess you only need to upload the HFEASY_WEB_UPGRADE file in the Upgrade SW menu. At least thats how it worked for me on my LPB130.

Lodhar, 2022-07-28:
Thanks for your answer! I will try that.

Lodhar, 2024-08-15:
I see it's still open. It worked but you need to flash with Edge (Firefox says it works but it doesn't). 
Go to http://[ip]/iweb.html
You need to use the "Upgrade application" not "Upgrade customized web page".
And upgrade with HFEASY_WEB_UPGRADE.bin

closed 2024-08-15 by Lodhar