ljalves/hfeasy #43: Retrieve device MQTT state after broker reboot

evil-life, 2023-01-17:
Hi,
I am not able to retrieve the state of the devices after the restart of the broker (Mosquitto running on Home Assistant).
Is there a specific command I can publish to force the devices publish their state? (Like it happens for Tasmota firmware).
I tried like this:
cmnd/Topic/POWER
with empty payload, but it doesn't work.
Thanks!

MrMusk, 2023-12-15:
Same problem here. I guess the device is not publishing its discovery payload as a retained message, and that's the reason why it changed the status to not available. I tried to fix this problem with a small workaround. I created an automation where Home Assistant checks if the status from the MQTT device is unavailable. If yes, it calls the service shell command: call_remote.

To implement this, add the following to your config.yaml file and change the IP address to that of your MQTT device:

`shell_command:
  call_remote: curl http://xxx.xxx.xxx.xxx:8080/save?restart=Restart`
  
I know it's late, but I hope it helps and the problem gets fixed in the firmware so that this 'workaround' becomes unnecessary.

evil-life, 2023-12-19:
Thanks for your reply, I already set that automation up, but I'm not very happy with it. Device restart will bring it to the initial state (on or off depending on your settings), so it could end up turning on the light in the bedroom in the middle of the night (happend to me) or turn off the kitchen lights while having dinner... not so comfortable :)
2 possible soulitions would be:
1) add the possibility to restore the last state after the restart;
2) add a command to force the device publish the actual state when HA restarts;
Thanks and regards