ljalves/hfeasy #44: Keeps showing up in Homa Assistant (OS) as unavailable until i reboot the plugs

PeteDenmark, 2023-02-02:
I have flashed with both the newest firmware and web interface, and both my HF-LPB100 plugs work great, and are auto discovered by Homa Assistant mosquitto broker.
But every time i restart HA, they are unavailable until i reboot the plugs.

phrogg, 2023-02-09:
I have the same issue.

PeteDenmark, 2023-02-10:
> I have the same issue.

I have made a workaround. I have made an automation that calls the HFeasy plugs reboot URL when HA starts. So HA starts, the plugs reboot, and everything is fine. Not the cleanest, but it works :)

You can call a URL with the Rest command:

So i configuration.yaml:

rest_command: !include rest_commands.yaml

Make a file called rest_commands.yaml anf put the following in it:

#restarts HFeasy plugs by calling their URLs
restart_hfeasy_1:
  url: 'http://192.162.1.150:8080/save?restart=Restart'
restart_hfeasy_2:
  url: 'http://192.168.1.151:8080/save?restart=Restart'
restart_hfeasy_3:
  url: 'http://192.168.8.152:8080/save?restart=Restart'
  
  You of course replace the url's with the url's for you specific plugs
  
  Then i go into automations and make one script with the RESTful command to call the rest_commands of each plug.
  
  And finally just make an automation with a trigger "when home assistant is started" and action and script whatever you called the script you made for the RESTful triggers.

phrogg, 2023-02-11:
> > I have the same issue.
> 
> I have made a workaround. I have made an automation that calls the HFeasy plugs reboot URL when HA starts. So HA starts, the plugs reboot, and everything is fine. Not the cleanest, but it works :)
> 
> You can call a URL with the Rest command:
> 
> So i configuration.yaml:
> 
> rest_command: !include rest_commands.yaml
> 
> Make a file called rest_commands.yaml and put the following in it:
> 
> ```
> #restarts HFeasy plugs by calling their URLs
> restart_hfeasy_1:
>   url: 'http://192.162.1.150:8080/save?restart=Restart'
> restart_hfeasy_2:
>   url: 'http://192.168.1.151:8080/save?restart=Restart'
> restart_hfeasy_3:
>   url: 'http://192.168.8.152:8080/save?restart=Restart'
> ```
> 
> You of course replace the url's with the url's for you specific plugs
> 
> Then i go into automations and make one script with the RESTful command to call the rest_commands of each plug.
> 
> And finally just make an automation with a trigger "when home assistant is started" and action and script whatever you called the script you made for the RESTful triggers.

Nice idea, unluckily unsuited for me as a reboot power cycles the plug.

PeteDenmark, 2023-02-12:
Yeah - that sucks.

Guess calling the HFeasy plugs on and off URLs with get commands is the only option left

on: http://\<deviceip\>/state?sw=1
off: http://\<deviceip\>/state?sw=0

sshaikh, 2024-09-04:
Is there any guidance on how to set this up with HA? I've used the following YAML:

```
# Switch
mqtt:
  - switch:
      unique_id: car_charger_smart_socket
      name: "Car Charger Smart Socket"
      state_topic: "homeassistant/switch/hfeasy_360D92LWT"
      command_topic: "homeassistant/switch/hfeasy_360D92LWT"
      payload_on: "ON"
      payload_off: "OFF"
```

But no luck.

Actually at rest I see the topic flicking from Online to Offline before the socket completely crashes - is the MQTT client in v.1.1 actually functional?

sshaikh, 2024-09-04:
Also, I can't tell where to publish the autodiscovery to? I feel that there are a whole bunch of mqtt settings I am missing.

sshaikh, 2024-09-08:
FYI I moved away from MQTT and went with the HTTP API instead:

https://github.com/ljalves/hfeasy/issues/47#issuecomment-2336763111