ljalves/hfeasy #50: Build instructions

stikonas, 2024-02-02:
It would be nice to have some README file with build instructions.

How do I build it for `HF-LPB100`? When I run `make` it only creates `LPB130` binaries...

```console
-----------------make target:HFEASY.bin 1st-dep:HFEASY.elf -----------------
/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-objcopy -O binary ./projects/build_lpxx30/HFEASY.elf ./projects/build_lpxx30/HFEASY.bin
/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-objdump -h ./projects/build_lpxx30/HFEASY.elf > ./projects/build_lpxx30/dump.log

BUILD COMPLETE :: LPB130 serial upload binary at: ./projects/build_lpxx30/HFEASY.bin
BUILD COMPLETE :: LPB130 web upload binary at: ./projects/build_lpxx30/HFEASY_UPGRADE.bin

make[1]: Leaving directory '/home/andrius/repositories/orvibo/hfeasy'
./tools/gen_upgrade_bin ./projects/build_lpxx30/HFEASY.bin ./projects/build_lpxx30/HFEASY_UPGRADE.bin ./tools/imgpkt
```

sshaikh, 2024-12-08:
Can you please explain how you even got that far? The project doesn't appear to build out of the box.

stikonas, 2024-12-08:
I installed arm-non-eabi cross-compiler on Gentoo (https://wiki.gentoo.org/wiki/Crossdev). Then `make` just worked.

stikonas, 2024-12-08:
> Can you please explain how you even got that far? The project doesn't appear to build out of the box.

I can't reproduce it right now though :(. In any case I was not able to build it for HF-LPB100. And probably not really worth spending too much time on these old sockets if you can buy any of the ESP32 (or at least ESP8266) devices and run Tasmota or ESPhome on them.

sshaikh, 2024-12-14:
Since I'm a glutton for punishment I've spent some time on this. I managed to get the Makefile to run on Arch using `gcc-arm-none-eabi-bin-7_2018_q2_update-1` (from the AUR). However I suspect the Makefile is a red herring - the repo seemed to move to "Keil MDK", a Windows based dev platform. That probably explains why there's no Makefile for LBP100. MDK project files can be found in ./project. 

MDK has limitations on its free version that only allows projects of 32k or less to be compiled. The good news is that there is a community licence that removes this restriction. The bad news is that the project is set up to use the Arm compiler v5, which doesn't recognise the community licence in this latest version of MDK. You can switch to v6, but then the code doesn't compile and would require some work above my paygrade to do so.

Since the Makefile worked, my next go is to return to Linux development and try to get a working Makefile for LBP100.

sshaikh, 2024-12-18:
So it looks like you can get MDK to compile this code via arm 5, and then link it using arm 6. Crazy but it works.